openhwfoundation/cva6 #2757: toolchain build error

RakeshpatilLB, 2025-02-03:
[image]

Got this error after executing "bash build-toolchain.sh $INSTALL_DIR"

jason23g, 2025-02-04:
Hello @RakeshpatilLB ,

Could you give the steps that you followed and got this error ?

Thanks in advance for your time

RakeshpatilLB, 2025-02-04:
@jason23g  i was trying to set up the repo in RHEL
Steps:(other than cloning the repo)
1) mkdir TOOLS
    RISCV= /home/TOOLS
    INSTALL_DIR=$RISCV

2) bash get-toolchain.sh

3) bash build-toolchain.sh $INSTALL_DIR

the given error occurred after executing the 3rd step. 
Thanks for responding, please let me know if u need more details to asses the problem.

jason23g, 2025-02-04:
Hello @RakeshpatilLB,

What version of GCC  are you using ?

RakeshpatilLB, 2025-02-05:
[image]
It is gcc 4.8.5 20150623.
If i am not mistaken , I think the version is the reason for the errors. And do we need preinstalled GCC before building the tool chain?

jason23g, 2025-02-07:
I think you should try GCC 13.1.0

RakeshpatilLB, 2025-03-10:
please let the issue be open for few more days, i havent figured out any solution regarding it. I want to share some more info regarding the issue in few days.

isa084, 2025-03-12:
@RakeshpatilLB , try using gcc 13.1.0 . 
gcc 4.8 is quite old.

closed 2025-04-15 by RakeshpatilLB