openhwfoundation/cva6 #2773: [BUG] SPIKE TANDEM failed in rv32ud-fmin test in riscv compliance tests.

jason23g, 2025-02-14:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hello guys,

I am using the spike tandem simulation and i observed that when i  use it in the test rv32ud-fmin.s the result of the co-simulation in the yaml after the check of the first two instructions are a little bit of strange. In the image below you can see clearly that the in third instruction which there is difference, the difference lies down in that the instruction in spike when executed the flag of trap is being enabled whereas in RTL in verilator is not being enabled. After checking with the simple simulation the correct execution of the test in spike, i observed that indeed the flag of the trap shouldn't have been enabled.  I am wondering if this is a known issue with the spike tandem or not  and if there is a solution.

Thanks in advance for your time and effort.

[image]

p.s.
The first file is the log of spike running standalone and the second is the result of the spike tandem simulation in yaml format.

[rv32ud-fmin.cv64a6_imafdc_sv39.log](https://github.com/user-attachments/files/18800292/rv32ud-fmin.cv64a6_imafdc_sv39.log)

[rv32ud-fmin.cv64a6_imafdc_sv39.log.yaml.txt](https://github.com/user-attachments/files/18800215/rv32ud-fmin.cv64a6_imafdc_sv39.log.yaml.txt)

JeanRochCoulon, 2025-02-16:
Spike-tandem supports only cv32a60x and cv32a65x.

jason23g, 2025-02-16:
Hello,

Yes I know it but i change the corresponding spike.yaml accordingly to the spike.yaml for the cv32a65x to be able to run spike tandem for cv64a6_imafdc. Is there anything else that i need to change in order to be able to support spike tandem for cv64a6_imafdc ?

JeanRochCoulon, 2025-02-17:
Today, the 60x and 65x Spike-Tandem need to be cleaned-up because Spike configuration comes from Spike.yaml but also from UVM (which is not good!!). The clean-up is planned by the end of March.

jason23g, 2025-02-17:
Ok, so if I wanted to do it and i have understood correctly, I would need firstly to configure spike only by the spike.yaml and not from UVM.

JeanRochCoulon, 2025-02-17:
Yes, it is better to configure by Spike.yaml

closed 2025-03-10 by jason23g