openhwfoundation/cva6 #2775: Question : How do I install gdb for debugging CVA6 on Genesys 2?

isa084, 2025-02-18:
Where is riscv64-unknown-elf-gdb (for use with OpenOCD) ? How do I install it ? I couldn't find any reference to it in the toolchain installed by scripts inside util/toolchain-builder.

I installed RISCV toolchain using the following steps : 
```
export RISCV=/path/to/riscv-toolchain-install-dir
INSTALL_DIR=$RISCV
bash get-toolchain.sh
bash build-toolchain.sh $INSTALL_DIR
```

 
To debug programs using gdb, I followed the two steps below : 

Step 1. Use OpenOCD  
`$ openocd -f fpga/ariane.cfg`

This worked as expected. I received the following output on my terminal : 

```
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
DEPRECATED! use 'adapter speed' not 'adapter_khz'
DEPRECATED! use 'adapter driver' not 'interface'
DEPRECATED! use 'ftdi vid_pid' not 'ftdi_vid_pid'
DEPRECATED! use 'ftdi channel' not 'ftdi_channel'
DEPRECATED! use 'ftdi layout_init' not 'ftdi_layout_init'
DEPRECATED! use 'ftdi layout_signal' not 'ftdi_layout_signal'
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Warn : `riscv set_prefer_sba` is deprecated. Please use `riscv set_mem_access` instead.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000014112f
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```

I opened another terminal and typed :
`riscv64-unknown-elf-gdb /path/to/elf` , assuming I had compiled an ELF file with riscv-none-elf-gcc

This command doesn't work because I did not have gdb installation on my system.

Let me know what to do here.
Thanks

Gchauvon, 2025-02-19:
It seems the part about GDB debugging in `README `has not been updated for newer version of GCC.
`riscv64-unknown-elf-gdb` is installed with older version of GCC. 

If you followed the installation in the README, your GDB is located in `YOUR_GCC_INSTALL_PATH/bin/riscv-none-elf-gdb`

isa084, 2025-02-19:
> If you followed the installation in the README, your GDB is located in YOUR_GCC_INSTALL_PATH/bin/riscv-none-elf-gdb

That is an issue. I do not see `riscv-none-elf-gdb` in the installation directory. Following are the contents of the directory :

<pre><font color="#26A269"><b>riscv-none-elf-addr2line</b></font>   <font color="#26A269"><b>riscv-none-elf-gcc-nm</b></font>      <font color="#26A269"><b>riscv-none-elf-nm</b></font>
<font color="#26A269"><b>riscv-none-elf-ar</b></font>          <font color="#26A269"><b>riscv-none-elf-gcc-ranlib</b></font>  <font color="#26A269"><b>riscv-none-elf-objcopy</b></font>
<font color="#26A269"><b>riscv-none-elf-as</b></font>          <font color="#26A269"><b>riscv-none-elf-gcov</b></font>        <font color="#26A269"><b>riscv-none-elf-objdump</b></font>
<font color="#26A269"><b>riscv-none-elf-c++filt</b></font>     <font color="#26A269"><b>riscv-none-elf-gcov-dump</b></font>   <font color="#26A269"><b>riscv-none-elf-ranlib</b></font>
<font color="#26A269"><b>riscv-none-elf-cpp</b></font>         <font color="#26A269"><b>riscv-none-elf-gcov-tool</b></font>   <font color="#26A269"><b>riscv-none-elf-readelf</b></font>
<font color="#26A269"><b>riscv-none-elf-elfedit</b></font>     <font color="#26A269"><b>riscv-none-elf-gprof</b></font>       <font color="#26A269"><b>riscv-none-elf-size</b></font>
<font color="#26A269"><b>riscv-none-elf-gcc</b></font>         <font color="#26A269"><b>riscv-none-elf-ld</b></font>          <font color="#26A269"><b>riscv-none-elf-strings</b></font>
<font color="#26A269"><b>riscv-none-elf-gcc-13.1.0</b></font>  <font color="#26A269"><b>riscv-none-elf-ld.bfd</b></font>      <font color="#26A269"><b>riscv-none-elf-strip</b></font>
<font color="#26A269"><b>riscv-none-elf-gcc-ar</b></font>      <font color="#26A269"><b>riscv-none-elf-lto-dump</b></font>
</pre>

isa084, 2025-02-21:
> I think you should be good by removing --disable-gdb in [util/toolchain-builder/config/global.sh](https://github.com/openhwgroup/cva6/blob/master/util/toolchain-builder/config/global.sh) and building the toolchain again.

I followed your suggestion and rebuilt the toolchain all over again (after deleting util/toolchain-builder/build,  util/toolchain-builder/src , $INSTALL_DIR). Following are the contents of the directory. I still do not see gdb here.
```
riscv-none-elf-addr2line   riscv-none-elf-gcc-nm      riscv-none-elf-nm
riscv-none-elf-ar          riscv-none-elf-gcc-ranlib  riscv-none-elf-objcopy
riscv-none-elf-as          riscv-none-elf-gcov        riscv-none-elf-objdump
riscv-none-elf-c++filt     riscv-none-elf-gcov-dump   riscv-none-elf-ranlib
riscv-none-elf-cpp         riscv-none-elf-gcov-tool   riscv-none-elf-readelf
riscv-none-elf-elfedit     riscv-none-elf-gprof       riscv-none-elf-size
riscv-none-elf-gcc         riscv-none-elf-ld          riscv-none-elf-strings
riscv-none-elf-gcc-13.1.0  riscv-none-elf-ld.bfd      riscv-none-elf-strip
riscv-none-elf-gcc-ar      riscv-none-elf-lto-dump
```

isa084, 2025-02-22:
@valentinThomazic , your suggestion worked !

I took 66caecd and then built the toolchain using default options. I can now see riscv-none-elf-gdb .
Then, I used some suggestions in this comment https://github.com/openhwgroup/cva6/issues/1803#issuecomment-2003829865 to execute a hello-world program completely. 

Coming back to the issue of gdb, does this need tp be fixed in (a) the main README.md, and (2) the gcc-toolchain folder ?

valentinThomazic, 2025-02-24:
We discussed with @math-gout who made the changes to the scripts and deleted the flags preventing gdb from being built.

valentinThomazic, 2025-02-25:
@isa084 If you consider https://github.com/openhwgroup/cva6/pull/2789 resolved the issue, please close it.

isa084, 2025-03-02:
@valentinThomazic , the issue has been resolved. Thanks !
Steps to reproduce : 
```

git clone https://github.com/openhwgroup/cva6.git cva6-issue-2775
cd cva6-issue-2775
git checkout dfdc72c

# Setup environment variables
export REPO_HOME=$(pwd)
export RISCV="${REPO_HOME}/util/toolchain-builder/riscv"
INSTALL_DIR=$RISCV
export NUM_JOBS=$(( $(nproc) * 2 / 3 )) # Set parallel jobs = 2/3 * total (logical) CPUs

# Install toolchain
cd ${REPO_HOME}/util/toolchain-builder
bash get-toolchain.sh && bash build-toolchain.sh $INSTALL_DIR

# check gdb installation : should display riscv-none-elf-gdb
ls -lrt $RISCV/bin/riscv* | grep gdb

```