openhwfoundation/cva6 #2794: [BUG] Compiler errors when running the cvxif_verif_regression

nachoge98, 2025-02-26:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

I'm going through the example implementation of a co-processor using the cvxif and I came across the following errors when running the regression for the existing co-processor tests (`bash verif/regress/cvxif_verif_regression.sh`):

```
Wed, 26 Feb 2025 16:59:38 INFO     Iteration number: 1
Wed, 26 Feb 2025 16:59:38 INFO     Processing regression test list : ../tests/testlist_cvxif.yaml, test: cvxif_add_nop
Wed, 26 Feb 2025 16:59:38 INFO     Found matched tests: cvxif_add_nop, iterations:1
Wed, 26 Feb 2025 16:59:38 INFO     CVA6 Configuration is  and target is cv64a6_imafdc_sv39
Wed, 26 Feb 2025 16:59:38 INFO     Compiling test: /home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S
Wed, 26 Feb 2025 16:59:38 INFO     /home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S: Assembler messages:
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:25: Error: unrecognized opcode `load_rs(a0, 0x332211)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:26: Error: unrecognized opcode `load_rs(a1, 0xDEADBEEF)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:27: Error: unrecognized opcode `load_rs(a2, 0xDEADBEEF)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:31: Error: too many positional arguments
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01010,01010,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:32:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01011,01010,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:33:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01010,01011,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:34:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:36: Error: unrecognized opcode `cus_add_rs1(01100,01010,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01010,01011,00000)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:38:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(00000,00000,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:47:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:48: Error: too many positional arguments
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:49: Error: too many positional arguments
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01010,01010,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:52:  Info: macro invoked from here
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:54: Error: unrecognized opcode `cus_add_rs1(01100,01010,01011)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:18: Error: illegal operands `r CUSTOM_3,0x1,0x0,(01010,01011,00000)'
/home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S:56:  Info: macro invoked from here

Wed, 26 Feb 2025 16:59:38 ERROR    ERROR return code: True/1, cmd:/home/ignaciogarciae/tools/cva6_tools/riscv-tools/bin/riscv-none-elf-gcc /home/ignaciogarciae/pdp_labs/cva6/verif/tests/custom/cv_xif/cvxif_add_nop.S           -I/home/ignaciogarciae/pdp_labs/cva6/verif/sim/dv/user_extension             -T../../config/gen_from_riscv_config/linker/link.ld -static -mcmodel=medany -fvisibility=hidden -nostdlib -nostartfiles ../tests/custom/common/syscalls.c ../tests/custom/common/crt.S -I../tests/custom/env -I../tests/custom/common -lgcc -o /home/ignaciogarciae/pdp_labs/cva6/verif/sim/out_2025-02-26/directed_tests/cvxif_add_nop.o  -march=rv64gc_zba_zbb_zbs_zbc -mabi=lp64d
```

nachoge98, 2025-02-27:
Just saw that @Gchauvon removed the missing definitions from cvxif_macros.h on a commit 3 weeks ago.
Any specific reason for it? I was planning on implementing my own co-processor and I wanted an initial dummy one working to modify.

Gchauvon, 2025-02-28:
We are currently finishing the verification of the CVXIF protocol implemented on CVA6.
We will update `cvxif_verif_regression.sh` to fix this error which is known on our side. It requires a bit of work around the verification infrastructure so that directed tests ([cvxif_verif_regression.sh](https://github.com/openhwgroup/cva6/blob/master/verif/regress/cvxif_verif_regression.sh))  and generated tests ([dv-generated-xif-tests.sh](https://github.com/openhwgroup/cva6/blob/master/verif/regress/dv-generated-xif-tests.sh)) compile and run with the same macros.