openhwfoundation/cva6 #2817: [BUG] <title>

jason23g, 2025-03-09:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hello guys,

I am trying to upgrade the Verilator version from 5.008 to 5.028. I changed the version number in both the verif/regress/install-verilator.sh script and in verif/sim/cva6.py, but when I run the smoke-tests.sh script, it fails to verilate. I am wondering if anything else needs to be done besides what I have already done, or if anyone has any idea why this is happening.

Thanks in advance for your time.

[verilator_error_5028.txt](https://github.com/user-attachments/files/19150611/verilator_error_5028.txt)

JeanRochCoulon, 2025-03-10:
By experience, we know that not all Verilator versions work for the cva6 projet (more due to Verilator weaknesses than cva6 RTL). That is the reason why we fixed the version to a functional one.

jason23g, 2025-03-10:
Okay, I understand. Could you let me know if you have tried any version of Verilator greater than or equal to 5.020?

valentinThomazic, 2025-03-12:
@jason23g iirc, we did not manage to use (consistently) Verilator versions greater than 5.008, which is why it is still the versions in the scripts.