openhwfoundation/cva6 #2852: [BUG] spike : Incorrect result values in instructions

AyoubJalali, 2025-03-21:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

After @zchamski implement the [new cvxif custom instructions](https://github.com/openhwgroup/core-v-verif/commit/620b4fa6639c88d42d18b3a02e635881184bd6af), now the issue is in the results of these instructions, and that affect also some other valid instructions results also.

zchamski, 2025-03-24:
@AyoubJalali, can you confirm that the issue is only observed on the 'master' branch of CVA6, or do you see it on the 'cv32a60x' branch as well ?

AyoubJalali, 2025-03-24:
> [@AyoubJalali](https://github.com/AyoubJalali), can you confirm that the issue is only observed on the 'master' branch of CVA6, or do you see it on the 'cv32a60x' branch as well ?

I think this is a core-v-verif issue, the problem is in CVV master

zchamski, 2025-03-24:
Yes it is is a CVV issue, and a duplicate of #2849 and it does not affect CV32A60X/CV32A65X anymore.  The discussion in #2849 explains why a temporary solution for CVA6 master is needed and why it is not just a CVV bump (CVV `master` got too far ahead of CVA6 `master`.)

closed 2025-03-27 by ASintzoff

ASintzoff, 2025-03-27:
For the record, fix thanks:

- PR core-v-verif 2595 https://github.com/openhwgroup/core-v-verif/pull/2595
- PR cva6 2837 https://github.com/openhwgroup/cva6/pull/2837