openhwfoundation/cva6 #2860: [BUG] No reference for Synthesis

deepeshsahoo, 2025-03-21:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

We need an updated manual for synthesis. There is absolutely no information provided currently.

JeanRochCoulon, 2025-03-22:
Right, there is no information because synthesis depends on CAD vendor tools and confidential techno. Open Source depends on contributors, feel free to add such information.

deepeshsahoo, 2025-03-25:
Hi Jean, 

I see a lot of errors around the HPDCACHE, especially when using design compiler. It fails to pick up a lot of files that I don't understand, for example, the HPDCACHE_DIR/rtl/include/hpdcache_typedef.svh

While I do understand that we don't want to share confidential technology, it would be good to know exactly what is being omitted or considered for synthesis.

RakeshpatilLB, 2025-04-15:
Provide the full path of the file that you want to include. This method worked for me.
Like this:

[image]