openhwfoundation/cva6 #2861: [TASK] Build ReadTheDocs documentation from `cv32a60x` branch

MikeOpenHWGroup, 2025-03-22:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

This task is complete when:

- [ ] the [CV32A60X documentation](https://docs.openhwgroup.org/projects/cva6-user-manual/07_cv32a60x/index.html) is generated from the `cv32a60x` branch.
- [ ] A documentation built is triggered by a pull-request.

### Required Changes

.

### Current Status

.

### Description of Done

.

MikeOpenHWGroup, 2025-03-22:
Hi @ASintzoff, at our 2025-03-20 meeting I took an Action to complete the sub-tasks above.  I am not 100% clear on what the team wants, so I have created this Issue to track the Task.

I have updated the RTD project to generate the [CVA6 documentation](https://docs.openhwgroup.org/projects/cva6-user-manual/index.html) from the `cv32a60x` branch.  Please have a look at the result.

I am not 100% certain on what we want here.  After my changes the above link will point to the most recent hash of either the `master` or `cv32a60x`branches.  I _think_ we want everything generated from the master branch, _except_ the [CV32A60X documentation](https://docs.openhwgroup.org/projects/cva6-user-manual/07_cv32a60x/index.html#cv32a60x-documentation) chapter, which should be generated from the head of the `cv32a60x` branch.  Is that correct?

ASintzoff, 2025-03-24:
This is the way to proceed I have in mind.

Before considering the first item is completed, I would prefer to have a merged PR on `cv32a60x` branch modifying the documentation. I have one in preparation (waiting internal review).

ASintzoff, 2025-03-25:
Unfortunately, the [CV32A60X documentation](https://docs.openhwgroup.org/projects/cva6-user-manual/07_cv32a60x/index.html#cv32a60x-documentation) chapter was not updated after merge of PR 2863.

For instance in section [3.1.4. Machine Implementation ID (mimpid) Register](https://docs.openhwgroup.org/projects/cva6-user-manual/07_cv32a60x/riscv/priv.html#_machine_implementation_id_mimpid_register) there is no more TODO in the source file but there is still one in the rendered HTML.

MikeOpenHWGroup, 2025-04-26:
Hi @ASintzoff, I _think_ this task is complete.  Can you confirm?

ASintzoff, 2025-04-28:
Sure, I close it as completed.

closed 2025-04-28 by ASintzoff