openhwfoundation/cva6 #2870: [TASK] CVA60X code coverage status

ASintzoff, 2025-03-26:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

This issue describes not covered lines and conditions.
The commit used for this description is [f4c26fc](https://github.com/openhwgroup/cva6/commit/f4c26fcc555c49289241dd3992fc12eb7a369bbd).

Line coverage

- load_store_unit.sv
`666        0/1     ==>          default: ;`

- compressed_decoder.sv
`324        0/1     ==>              illegal_instr_o = 1'b1;`
`932        0/1     ==>              illegal_instr_o = 1'b1;`

- decoder.sv
`441        0/1     ==>              default: illegal_instr = 1'b1;`

- scoreboard.sv
`189        1/2     ==>        if (mem_q[i].sbe.fu == ariane_pkg::NONE && mem_q[i].issued) mem_n[i].sbe.valid = 1'b1;`

Condition coverage

- csr_regfile.sv

```
 LINE       2395
 SUB-EXPRESSION (((!1'b0)) && (csr_op_i == WFI) && ((!ex_i.valid)))
                 ----1----    --------2--------    -------3-------
-   1   0   Not Covered
```

- serdiv.sv
```
 LINE       237
 SUB-EXPRESSION (op_a_sign ^ op_b_sign ^ op_b_neg_one)
                 ----1----   ----2----   ------3-----
0   0   1   Not Covered
1   0   1   Not Covered
```
- frontend.sv
```
 LINE       635
 EXPRESSION (resolved_branch_i.is_mispredict & (resolved_branch_i.cf_type == JumpR))
             ---------------1---------------   ------------------2-----------------
0   1   Not Covered
```
```
 LINE       705
 EXPRESSION (halt_i ? '0 : ({{(32'b00000000000000000000000000100000 - 3) {1'b0}}, 3'b100}))
             ---1--
1   Not Covered
```

- instr_scan.sv (to be completed: 16 conditions not covered in instance 1 but such conditions are covered in instance 0)

- cva6_fifo_v3.sv
```
 LINE       99 (only i_fifo_address)
 EXPRESSION (push_i && ((~full_o)))
             ---1--    -----2-----
1   0   Not Covered
```
```
 LINE       119 (only gen_instr_fifo[1].i_fifo_instr_data)
 EXPRESSION (pop_i && ((~empty_o)))
             --1--    ------2-----
1   0   Not Covered
```
```
 LINE       131
 EXPRESSION (push_i && pop_i && ((~full_o)) && ((~empty_o)))
             ---1--    --2--    -----3-----    ------4-----
1   1   1   0   Not Covered
```

- decoder.sv
```
 LINE       1464
 EXPRESSION (((~ex_i.valid)) && (is_illegal_i || illegal_instr))
             -------1-------    ---------------2---------------
0   1   Not Covered
```
```
 LINE       1464
 SUB-EXPRESSION (is_illegal_i || illegal_instr)
                 ------1-----    ------2------
1   0   Not Covered
```

- issue_read_operands.sv
```
 LINE       251
 EXPRESSION (x_issue_valid_o && x_issue_ready_i && x_issue_resp_i.accept)
             -------1-------    -------2-------    ----------3----------
0   1   1   Not Covered
1   0   1   Not Covered
```

- rr_arb_tree.sv
``` LINE       267
 EXPRESSION (gen_arbiter.req_d[(1 * 2)] | gen_arbiter.req_d[((1 * 2) + 1)])
             -------------1------------   ----------------2---------------
0   1   Not Covered
```
```
 LINE       270
 SUB-EXPRESSION (gen_arbiter.req_d[((1 * 2) + 1)] & gen_arbiter.rr_q[((gen_arbiter.NumLevels - 1) - 3)])
                 ----------------1---------------   -------------------------2-------------------------
1   0   Not Covered
```

- scoreboard.sv
```
 LINE       189
 EXPRESSION ((mem_q[i].sbe.fu == NONE) && mem_q[i].issued)
             ------------1------------    -------2-------
1   1   Not Covered
```

### Required Changes

- explain why these lines/conditions are not covered

### Current Status

- 5 lines not covered
- 31 conditions not covered

### Description of Done

Remaining not covered lines/conditions are justified

ASintzoff, 2025-03-31:
5 lines cannot be covered
 
- load_store_unit.sv
`666        0/1     ==>          default: ;`
Cannot be covered (as data_misaligned is equal to 1 only for LOAD or STORE)

- compressed_decoder.sv
`324        0/1     ==>              illegal_instr_o = 1'b1;`
Cannot be covered (as all 8 possible values are already addressed outside the default)
`932        0/1     ==>              illegal_instr_o = 1'b1;`
Cannot be covered (as all 8 possible values are already addressed outside the default)

- decoder.sv
`441        0/1     ==>              default: illegal_instr = 1'b1;`
Cannot be covered (as all 8 possible values are already addressed outside the default)

- scoreboard.sv
`189        1/2     ==>        if (mem_q[i].sbe.fu == ariane_pkg::NONE && mem_q[i].issued) mem_n[i].sbe.valid = 1'b1;`
Cannot be covered
`mem_q[i].sbe.fu == ariane_pkg::NONE` is always `0` as in `decoder.sv`, `fu` is never `ariane_pkg::NONE` as `frontend.sv/instr_queue.sv` never sets `ex.valid` to `1` because there is not exception raised by frontend in cv32A60x configuration (no PMP, no MMU: no INSTR_ACCESS_FAULT, no FE_INSTR_PAGE_FAULT)

ASintzoff, 2025-03-31:
Conditions cannot be covered
 
- serdiv.sv
```
 LINE       237
 SUB-EXPRESSION (op_a_sign ^ op_b_sign ^ op_b_neg_one)
                 ----1----   ----2----   ------3-----
0   0   1   Not Covered
1   0   1   Not Covered
```
Cannot be covered without code modification (see not merged PR #2857)
line 104: `op_b_neg_one = lzc_b_no_one & op_b_sign;` then when `op_b_sign == 0`, `op_b_neg_one == 0`

ASintzoff, 2025-03-31:
Conditions covered at least in one module instance

`instr_scan.sv`: all conditions are covered in instance 0. In instance 1, 16 conditions are not covered because this instance receives only compressed instructions

`cva6_fifo_v3.sv`: all conditions are covered in `gen_instr_fifo[0].i_fifo_instr_data` instance.

ASintzoff, 2025-04-01:
Conditions not covered in external module

`pulp_platform/common_cells/rr_arb_tree.sv`: 9 conditions not covered in the 3 instances

ASintzoff, 2025-04-01:
Conditions covered elsewhere in the same module
 
- decoder.sv
```
 LINE       1464
 EXPRESSION (((~ex_i.valid)) && (is_illegal_i || illegal_instr))
             -------1-------    ---------------2---------------
0   1   Not Covered
```
```
 LINE       1464
 SUB-EXPRESSION (is_illegal_i || illegal_instr)
                 ------1-----    ------2------
1   0   Not Covered
```
These conditions are covered at lines 1586 and 1598.

ASintzoff, 2025-04-18:
All remaining holes are explained and waived.

closed 2025-04-18 by ASintzoff

MikeOpenHWGroup, 2025-04-26:
Hi @ASintzoff, I do not fully understand why the following cannot be covered:

> * serdiv.sv
>
> ```
>  LINE       237
>  SUB-EXPRESSION (op_a_sign ^ op_b_sign ^ op_b_neg_one)
>                  ----1----   ----2----   ------3-----
> 0   0   1   Not Covered
> 1   0   1   Not Covered
> ``` [...]