openhwfoundation/cva6 #2871: [TASK] CV32A60X TODO, TBD, FIXME in RTL comments status

ASintzoff, 2025-03-26:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

This issue describes TODO, TBD, FIXME.
The commit used for this description is [f4c26fc](https://github.com/openhwgroup/cva6/commit/f4c26fcc555c49289241dd3992fc12eb7a369bbd).

The considered RTL files are located in:
`core/` , `core/frontend/` and `core/include/` directories.

### Required Changes

- explain why these TODO, TBD, FIXME are harmless

### Current Status

- 27 TODO
- 36 FIXME

### Description of Done

Remaining TODO, TBD, FIXME are justified

yanicasa, 2025-04-01:
some FIXME/TODO are being deleted or renamed in IMPROVEMENT (PR not yet open):

Please find the status of the remaining one:

File | Description | Status for cv32a60x
-- | -- | --
amo_buffer.sv | OBI paritiy signal | not in scope
commit_stage.sv | fence.i + dcache in WB | not in scope
constroller.sv -136 | fence.i + dcache in WB | not in scope
constroller.sv -236 | PERF optim:  PMA allow to not   flush_bp  to remove speculative check   to virtual mem | not in scope
csr_regfile - 1340 | implement hgeie write | not in scope
cva6_pipeline.sv 119 | OBI mmu_ptw/zcmt bus use old bus | not in scope
cva6_pipeline.sv 811 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 865 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 866 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 965 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 996 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1066 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1127 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1136 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1137 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1176 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1205 | Old writeback cache parameter to perf counter | not in scope
cva6_pipeline.sv 1222 | Perf counter if_empty is badly calculated in superscalar | not in scope
cva6_pipeline.sv 1238 | Old writeback connection to perf counter | not in scope
cva6_pipeline.sv 1274 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1289 | Old accelerator connection (useless) | not in scope
cva6_pipeline.sv 1289 | Old accelerator connection (useless) | not in scope
cva6.sv | 15 fixme | not in scope
cvxif_issue_register_commit_if | Always commit, no speculationin exec stage? |  
decoder.sv -195 | rtype opcode get rs2 from rtype/itype structure? |  
decoder.sv -545 | related to FPU | not in scope
decoder.sv -1342 | code cleaning for AMO | not in scope
exec_stage.sv | Superscalar + RVH | not in scope
instruction realign.sv -140 | just a comment? In case of fetch width ==64 | not in scope
instruction realign.sv -253 | just a comment? In case of fetch width ==64 | not in scope
issue_read_operand.sv | FLU + Superscalar | not in scope
store_buffer.sv | OBI paritiy signal |  
cvxif_example_copro.sv | copro example | not in scope
cvxif_instr_pkg.sv | copro example | not in scope
instr_queue.sv -302 | Superscalar + code cleaning | not in scope
build_config_pkg.sv -59 | MEM_TID_WIDTH Hpdcache require same size for transaction   id  input and output | not in scope????
riscv_pkg.sv -23 | Delete XLEN PLEn from the package when ref in ccv and hpdache   no more use it (code cleaning) | not in scope????

yanicasa, 2025-04-10:
https://github.com/openhwgroup/cva6/commit/11ce298969d900bd99be35c5894459c756815bc7 remove some TODO

yanicasa, 2025-04-10:
All todo justified for cv32a60x

ASintzoff, 2025-04-14:
All remaining TODO are explained and waived.