openhwfoundation/cva6 #2876: How to port this ariena CPU to other system

torettoluke, 2025-03-27:
How to port this ariena CPU to other system？
Which file are necessary？ Do you have some suggestion about porting?

WORMMMMMM, 2025-03-31:
I want to learn about this stuff too. I 'm trying to replace zeroriscy core with this cva6 core. If you have any progress ,please contact me.

OlivierBetschi, 2025-05-06:
Hello, the filelist is available in the Flist [file](https://github.com/openhwgroup/cva6/blob/master/core/Flist.cva6). You can either use an existing configuration (highly recommended, [cv32a60x)](https://github.com/openhwgroup/cva6/blob/master/core/include/cv32a60x_config_pkg.sv) is I think the most verified) or define your own (see in https://github.com/openhwgroup/cva6/tree/master/core/include for examples).

JackySu, 2025-05-09:
> Hello, the filelist is available in the Flist [file](https://github.com/openhwgroup/cva6/blob/master/core/Flist.cva6). You can either use an existing configuration (highly recommended, [...]

Hello! may I ask if there is any specific commit or tag I can checkout on to run smoke-tests in verif/regress/smoke-tests*.sh ? I checkout on tags like v5.6.0 v6.0.0 etc. and they all failed with reasons like illegal instructions, or just didn't manage to compile at all. 

I followed all instructions in README and installed verilator and spike required, yet I didn't make it work. Would really like to get help from you :<

OlivierBetschi, 2025-05-09:
Hi @JackySu. I would expect the recent tags and the latest master to successfully pass the smoke-tests. Ideally start from the latest master branch, make sure your compiler is properly set and simulator as well. If you cannot debug the issue, I would suggest you to create a new ticket for that, it makes things easier to track.