openhwfoundation/cva6 #2878: [mmu] How is access flag managed in mmu

khandelwaltanuj, 2025-03-27:
Hello,

I am working on a bench of MMU and I am trying to understand how the bit "a" of page table entry is managed. MMU seems to flag an access fault if a bit is not set. 

The ISA documentations says :
_**Two schemes to manage the A and D bits are defined:

The Svade extension: when a virtual page is accessed and the A bit is clear, or is written and the D bit is clear, a page-fault exception is raised.

When the Svade extension is not implemented, the following scheme applies.

When a virtual page is accessed and the A bit is clear, the PTE is updated to set the A bit. When the virtual page is written and the D bit is clear, the PTE is updated to set the D bit. When G-stage address translation is in use and is not Bare, the G-stage virtual pages may be accessed or written by implicit accesses to VS-level memory management data structures, such as page tables.**_

It seems to me that the svade extension is not implemented within the CVA6. 
I am not sure if I understand well what happens in the case when SVADE is not implemented. 

MMU reply with an access fault and software should update the PTE before doing the translation ? 

Thanks and 
Regards
Tanuj Khandelwal

NicolasDerumigny, 2025-03-27:
My understanding is that when Svade is not enabled, the update of access / dirty bit is done in hardware. However, when Svadu extension is not supported, then the system behaves as if Svade was **enable**, i.e. case 1 (page fault on mismatching access / dirty bit). I believe CVA6 does not implement Svadu and expect the software to handle D/A bits, but I may be wrong - I'm just a user, not a designer.

khandelwaltanuj, 2025-03-28:
Thanks Nicolas,

Thats what I am observing. 

Regards
Tanuj Khandelwal

closed 2025-04-10 by khandelwaltanuj