openhwfoundation/cva6 #2890: [pmp access] instruction passes even if x bit is set to 0

khandelwaltanuj, 2025-04-01:
Hello,

I am running a test of mmu with instruction fetches and I observed that translation completes without any fault even if x bit of pmp cfg register is set to 0. 

I am running the test in privilege mode S with pmpcfg set as 01b. 

It only gives an error when r bit is set to 0. 

I am wondering if its normal to not have an instruction access fault when x=0 ? 

Thanks and Regards
Tanuj Khandelwal

MikeOpenHWGroup, 2025-04-01:
Hi @khandelwaltanuj, the Privileged spec (20211203) is pretty clear about this.  In section 3.7.1, p. 58:
> Attempting to fetch an instruction from a PMP region that does not have execute permissions raises
an instruction access-fault exception.

Of course, this should not affect the _translation_ performed by the MMU, nor should it block the memory fetch transaction from occurring.

khandelwaltanuj, 2025-04-01:
Hi @MikeOpenHWGroup 

I have couple of questions here: 

ISA as you pointed out if x bit of pmpcfg == 0, it shall give an instruction access-fault. 
1. I do not see the instruction access-fault in the case where when x == 0 
2. Normally in the cases where there is page fault or access fault, the translation are not performed(we have just offset as paddr. 

Here everything seems to be passing  correctly. And I was expected an instruction access fault. 

Thanks and regards
Tanuj

khandelwaltanuj, 2025-04-01:
I have an example here:
where with pmpcfg = x18 I observe an instruction fault and translation is stopped at the first level of PTE. 

And with pmpcfg = x19 (r == 1, x== 0) translation is performed without any instruction fault and 3 levels of PTE are accessed

#WITH PPM x18
#
# UVM_INFO 9153000 ps: reporter [MMU SB ICACHE REQ] Fetch Req=1(x), Fetch Vaddr=b09c008bb(x)
# UVM_INFO 9155500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM REQ READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0ec5800(x) DATA       = e65f7a8a0e461(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM RSP READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0ec5800(x) DATA       = e65f7a8a0e461(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE ENTRY RCV] Fetch Addr = b09c008bb(x)  addr=e65f7a8a0ec5800(x), PPN = 3997dea2839 RSW = 0 D   = 0 A   = 1 G   = 1 U   = 0 X   = 0 W   = 0 R   = 0 V   = 1 
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PPNs RCV] Fetch Addr = b09c008bb(x) PPN NUMBER=0(d), PPN=39(x)
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PPNs RCV] Fetch Addr = b09c008bb(x) PPN NUMBER=1(d), PPN=114(x)
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PPNs RCV] Fetch Addr = b09c008bb(x) PPN NUMBER=2(d), PPN=e65f7a(x)
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PMP ACCES STATUS] Status = 0(d) Fetch_addr = b09c008bb(x)  Mem Addr= 3997dea283b160(x), Base Addr = 3997dea2800000(x), Max Addr=3997dea2880000(x) PMPCFG = 18(x)
# UVM_INFO 9156500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE LEVEL] Fetch Addr = b09c008bb(x) PTE LEVEL=3(d) leaf=0(x)
# UVM_INFO 9158000 ps: reporter [MMU SB ICACHE RSP EXCEPTION] Cause=1, Tval=b09c008bb(x), Tval20(x), Tinst=0(x) GVA=0(x), Valid=1(x)
# UVM_INFO 9158000 ps: uvm_test_top.env.m_mmu_sb [MMU SB ITLB MISS] Fetch Addr = b09c008bb(x) EXP PADDR= 8bb(x)
# UVM_INFO 9158000 ps: uvm_test_top.env.m_mmu_sb [MMU SB FETCH PADDR INFO] Fetch Addr = b09c008bb(x), VPN = b09c00(x), Rcv Addr = 8bb(x) Exp Addr= 8bb(x)
#
#
# WITH PMPCFG x19
#
# UVM_INFO 17221000 ps: uvm_test_top.env.icache_agent.monitor [GENERIC UVM MON] Addr= b09c008bb(x)
# UVM_INFO 17221000 ps: reporter [MMU SB ICACHE REQ] Fetch Req=1(x), Fetch Vaddr=b09c008bb(x)
# UVM_INFO 17223500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE LEVEL] Fetch Addr = b09c008bb(x) SATP = 3997dea283b(x) PTE LEVEL=3(d)
# UVM_INFO 17223500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM REQ READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0ec5800(x) DATA       = e65f7a8a038e1(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 17224500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM RSP READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0ec5800(x) DATA       = e65f7a8a038e1(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 17224500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE ENTRY RCV] Fetch Addr = b09c008bb(x)  addr=e65f7a8a0ec5800(x), PPN = 3997dea280e RSW = 0 D   = 1 A   = 1 G   = 1 U   = 0 X   = 0 W   = 0 R   = 0 V   = 1 
# UVM_INFO 17224500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PMP ACCES STATUS] Status = 1(d) Fetch_addr = b09c008bb(x)  Mem Addr= 3997dea283b160(x), Base Addr = 3997dea2800000(x), Max Addr=3997dea2880000(x) PMPCFG = 19(x)
# UVM_INFO 17226500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE LEVEL] Fetch Addr = b09c008bb(x) SATP = 3997dea283b(x) PTE LEVEL=2(d)
# UVM_INFO 17226500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE ENTRY] Fetch Vaddr = b09c008bb(x) addr=e65f7a8a0ec5800(x), PPN = 3997dea280e RSW = 0 D   = 1 A   = 1 G   = 1 U   = 0 X   = 0 W   = 0 R   = 0 V   = 1 
# UVM_INFO 17226500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM REQ READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0389c00(x) DATA       = e65f7a8a000a1(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 17227500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM RSP READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0389c00(x) DATA       = e65f7a8a000a1(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 17227500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE ENTRY RCV] Fetch Addr = b09c008bb(x)  addr=e65f7a8a0389c00(x), PPN = 3997dea2800 RSW = 0 D   = 1 A   = 0 G   = 1 U   = 0 X   = 0 W   = 0 R   = 0 V   = 1 
# UVM_INFO 17227500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PMP ACCES STATUS] Status = 1(d) Fetch_addr = b09c008bb(x)  Mem Addr= 3997dea280e270(x), Base Addr = 3997dea2800000(x), Max Addr=3997dea2880000(x) PMPCFG = 19(x)
# UVM_INFO 17229500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE LEVEL] Fetch Addr = b09c008bb(x) SATP = 3997dea283b(x) PTE LEVEL=1(d)
# UVM_INFO 17229500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PTE ENTRY] Fetch Vaddr = b09c008bb(x) addr=e65f7a8a0389c00(x), PPN = 3997dea2800 RSW = 0 D   = 1 A   = 0 G   = 1 U   = 0 X   = 0 W   = 0 R   = 0 V   = 1 
# UVM_INFO 17229500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM REQ READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0000000(x) DATA       = e65f7a8a000ef(x) ERR        = 0(x) EX ERR     = 0(x) 
# UVM_INFO 17230500 ps: uvm_test_top.env.m_mmu_sb [MMU SB MEM RSP READ] Fetch Addr = b09c008bb(x) ID         = 0(x) ADDR       = e65f7a8a0000000(x) DATA       = e65f7a8a000ef(x) ERR        = 0(x) EX ERR     = 0(x) 
[... cut ...]
# UVM_INFO 17230500 ps: uvm_test_top.env.m_mmu_sb [MMU SB PMP ACCES STATUS] Status = 1(d) Fetch_addr = b09c008bb(x)  Mem Addr= 3997dea2800000(x), Base Addr = 3997dea2800000(x), Max Addr=3997dea2880000(x) PMPCFG = 19(x)
# UVM_INFO 17230500 ps: uvm_test_top.env.m_mmu_sb [MMU SB LEAF PTE] Fetch Addr = b09c008bb(x) PTE=e65f7a8a000ef(x) PPN=3997dea2800(x)
# UVM_INFO 17234000 ps: uvm_test_top.env.m_mmu_sb [MMU SB ITLB MISS] Fetch Addr = b09c008bb(x) EXP PADDR= 3997dea28008bb(x)
# UVM_INFO 17234000 ps: uvm_test_top.env.m_mmu_sb [MMU SB FETCH PADDR INFO] Fetch Addr = b09c008bb(x), VPN = b09c00(x), Rcv Addr = 3997dea28008bb(x) Exp Addr= 3997dea28008bb(x)

MikeOpenHWGroup, 2025-04-01:
You are right, this looks suspicious to me.  I have not been directly involved in the development of the CVA6 verification environment, so I am not the best person to ask about this.  Also, you are breaking new ground as I doubt very much that this was extensively tested in the CVA6 up to this point (as you know the CV32A60X does not support PMP).

I am starting to suspect this is an RTL bug.

Also, I cannot find the mmu_sb anywhere - are you running in a local verification environment?

khandelwaltanuj, 2025-04-01:
I am sorry this is not the correct code. I will look for the correct lines.

khandelwaltanuj, 2025-04-02:
Hi @MikeOpenHWGroup 

I see that there is pmp instantiated in load_store_unit. It makes me wonder if the one in mmu is not redundant ? 
https://github.com/openhwgroup/cva6/blob/7b3054156eb996b9f6b4ba79403b65fee2a1d64c/core/load_store_unit.sv#L370

Regards
Tanuj Khandelwal

khandelwaltanuj, 2025-04-08:
Hi @AngelaGonzalezMarino 

I had open another issue here. Within the PTW of mmu PMP is instatiated assuming all access are READ, which, in my opinion, probably is not correct. 

Regards
Tanuj

AngelaGonzalezMarino, 2025-04-08:
I don't know about this, maybe @ninolomata can help here?

AngelaGonzalezMarino, 2025-04-08:
I found this in the MMU documentation:

In case of a DTLB miss, potential exceptions are monitored during the page table walk. If the PTW indicates a page fault, the corresponding page fault related to the requested type is signaled. If the PTW indicates an access exception, the load access fault is indicated through address translation because the page table walker can only throw load access faults.

ninolomata, 2025-04-08:
Hi @AngelaGonzalezMarino, i have noticed there's a missing PMP check on the MMU. It looks like it was removed in an earlier commit, and I don’t see it being handled elsewhere.

ninolomata, 2025-04-08:
Hi @khandelwaltanuj,

> As per my understanding there are multiple checks missing.
> 
>     1. It doesnt check for instruction
> 
>     2. It doesnt do check in Machine mode
> 
>     3. It doesnt do any check in case of a write access ?

1. I had a closer look at the MMU (sorry, it's been a while since I last checked the RTL), and it looks like the PMP checks for both data and instruction accesses have been moved to the load-store unit. Does your setup includes load_store_unit ? 
2. Could you clarify which specific scenarios you're referring to?
3. If you're talking about accesses within the PTW, then it does not check for write accesses. it makes sense that these are READ accesses only, since the MMU is just reading PTEs. CVA6 doesn’t update the A and D flags directly. Instead, it traps to handle that. The PMP checks on the translated address should respect the access type, and from what I have seen, that’s already being handled correctly.

khandelwaltanuj, 2025-04-08:
Hi @ninolomata 

My setup has mmu stand alone, so no load store unit. I may include pmp_data_if in my bench.  

I looked at the load store unit. And I am not convinced if it covers everything. 
It seems to performe checks only on translated addresses but not on accessing PTE. 

As per my understading of Priviliged ISA, it should performe all PMP checks at every level of PTE. 

Following is a snippet from the ISA specification. (10.3. Sv32: Page-Based 32-bit Virtual-Memory Systems | Page 109) 

2. Let pte be the value of the PTE at address a+va.vpn[i]×PTESIZE. (For Sv32, PTESIZE=4.) If
accessing pte violates a PMA or PMP check, raise an access-fault exception corresponding to the
original access type.

I may be totally wrong, I am not totally convinced though that PTW shall only do READ access checks on PTE. And I am also not convinced that PMP/PMA checks are applied only on translated addresses. 

Thanks and Regards
Tanuj Khandelwal

ninolomata, 2025-04-08:
If you look closely at the PTW module—specifically [here](https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_ptw.sv#L232C3-L244C5)—you’ll see there’s a separate PMP check that handles PTE access violations. This is used by the MMU exclusively for reading PTEs, and there are no writes performed here.

On the other hand, the PMP checks in the LSU handles violations on the translated addresses, and that can be either a read or a write, depending on whether it’s a load or a store.

khandelwaltanuj, 2025-04-08:
I have a feeling that my understanding of PMP is not aligned with yours. 

As per my understanding every access that are done for a PTE read are checked against PMPCFG (x, r, w) fields.

For me if I am doing an instruction fetch, the address to read every level of PTE shall satisfy the PMPAddr range and PMPCfg access conditions. Let me know if that is not the case please ? 

@MikeOpenHWGroup do you have any comment on this? I would not mind a third opinion because my understand of ISA seems to suggest if any accessing PTE voilates PMP/PMA check, it shall raise a fault.  

@ninolomata seems to suggest that it is done only for translated address.

MikeOpenHWGroup, 2025-04-10:
Hi @fatimasaleem, can you copy your comments from [Mattermost](https://mattermost.openhwgroup.org/all-users/pl/geeco6hzmtb7mcntgpeg57wgqy) here so that they are not lost?  Thanks!

fatimasaleem, 2025-04-11:
Hi @khandelwaltanuj 

Reading the PTEs from the memory is independent from the original request, and we only need read permission for this, and the final leaf PTE holds the permissions for the original request.
If the original request was an instruction fetch, the PTEs address should have the read access grant from PMP (x is not required). And finally when we will have the leaf PET, that should have pte.x bit set, so that we can generate the final physical address. That physical address then send out to PMP where we check both R and X permissions for that final address. 

But please note that if accessing pte violates a PMP check (no read permission was given), then the access-fault exception will be raised according to the original access type (i.e, instruction access fault in our example).

Following are the spec references for clarity.

[image]

[image]

khandelwaltanuj, 2025-04-11:
Thanks a lot @fatimasaleem 

Regards
Tanuj Khandelwal

closed 2025-04-11 by khandelwaltanuj