openhwfoundation/cva6 #2894: [BUG] pmp does not support NA4 configuration

khandelwaltanuj, 2025-04-01:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hello @JeanRochCoulon, @MikeOpenHWGroup 

I observed that following commits have removed NA4 case from the rtl of pmp_entry.sv. It means NA4 is not supported anymore in CVA6 pmp. 

https://github.com/openhwgroup/cva6/commit/2155d0e9c4a54998f8bb9aa6c28eb84b2a55b56f

Any reason for doing this ? 

Thanks and Regards
Tanuj Khandelwal

JeanRochCoulon, 2025-04-01:
NA4 has never been supported by CVA6. Lines dedicated to NA4 were present in the RTL, but they were dead lines. That is the reason why I removed thme: to increase code coverage and to provide a cleaner code.

khandelwaltanuj, 2025-04-02:
Hello,

I am little surprised to hear that. The privileged ISA has a very clear definition about it. And it was very well coded in the RTL. 
Why would CVA6 decide not to support it ? 

Regards
Tanuj Khandelwal

JeanRochCoulon, 2025-04-02:
It is not a decision, NA4 has never been functional. Wait for volunters to add it !

khandelwaltanuj, 2025-04-02:
Thanks a lot for the explanation. 

Regards
Tanuj

closed 2025-04-03 by khandelwaltanuj