openhwfoundation/cva6 #2903: in mmu lsu unit response valid (lsu_valid_o) remains high for 2 clock cycles

khandelwaltanuj, 2025-04-07:
Hello, 

During my verification of MMU, I am observing that for LSU, the réponse lsu_valid_o remains high for 2 clock cycles. wondering if anyone has any idea if it is normal ? By the way there is no ready signal in MMU. The signal req (kind of valid) has to remain high until the response is received. 

Thanks and Regards
Tanuj

MikeOpenHWGroup, 2025-04-07:
Hi @khandelwaltanuj, can you be more specific about the signal you are discussing here?  Is it [core/cva6_mmu/cva6_mmu.sv](https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_mmu.sv#L63)?  If so, perhaps @AngelaGonzalezMarino can comment.

khandelwaltanuj, 2025-04-08:
Hi 

Yes it is lsu_valid_o as you pointed out Mike. 

Regards
Tanuj Khandelwal

AngelaGonzalezMarino, 2025-04-08:
Hi @khandelwaltanuj, are you talking about the case when there is a dtlb hit?
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_mmu.sv#L568C1-L569C28
If so, lsu_valid_o remains high as long as there is a hit, and the req signal is still active. 
Are you testing the mmu "alone" or running the tests on cva6?
Inside cva6 it is possible that it takes 2 cycles to change the request, but I am not sure if that is (or should be) always the case.

khandelwaltanuj, 2025-04-08:
Hi @AngelaGonzalezMarino 

I am testing MMU standalone. What I am seeing is that lsu_valid_o remains high for another cycle once lus_req_i goes low. 
As soon as I do back to back request, it just breaks down. 

I will try to update a screen shot. I hope it works. 
[image]

For me all the sequential logic added here seems problematic. 
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_mmu.sv#L732

Thanks and Regards
Tanuj Khandelwal

AngelaGonzalezMarino, 2025-04-08:
Hi,
The delay (i.e. the registers) is there to sync with the ptw in case of a miss (and/or error/exception). 
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/docs/03_cva6_design/_static/ptw_state_diagram.png
This https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_mmu.sv#L641can't be evaluated until the ptw_error signal is propagated.
That is why the valid is sent one cycle after the hit.

khandelwaltanuj, 2025-04-08:
But they delay makes the lsu_valid_o, dtlb_hit_q, etc remains high for one extra clock cycles after lsu_req_i 

I am uploading another screenshot here. Can you please confirm me what you are seeing is normal ? The signal lsu_valid_o (in green) remains high for another clock cycles after lsu_req_i (in yellow). For me the back to back request cannot work here.

[image]

khandelwaltanuj, 2025-04-08:
Please see here the a case of  back to back response. 

there is lsu_vaddr_i: 29D5713FCD followed by 15849e011F followed by 3D63FBE118. 
As per my understanding of the waves, either 15849e011F does not have any response or has the same response as  29D5713FCD. It looks very suspicious to me. Except if you tell that back to back requests are not allowed. 

[image]

Regards
Tanuj Khandelwal

khandelwaltanuj, 2025-04-08:
By the there are no such issues in the case of icache interface. There are no registers to sync.

AngelaGonzalezMarino, 2025-04-08:
I think I understand your point now, I have the feeling that the reason for different behaviour between instruction and data is how the requests are handled (interfaces are different, and the mmu is adapted to that).
But I don't know if back to back request is a realistic use case or not.
@ninolomata maybe you can help here too? Thanks!

khandelwaltanuj, 2025-04-08:
Hi @cfuguet 

I allow myself to add César in the discussion, who I think is an active user of CVA6 and mmu. 

César, we are discussing an case  in the mmu which do not seems to allow back to back lsu requet. And I would not mind your opinion on that. 

I also allow myself also to say that "is it a not realistic use case" may not be a valid statement. 
We may rather say that it is not acceptable and we modify the specs to precise this condition. 

thanks for your response
Regards
Tanuj Khandelwal

cfuguet, 2025-04-08:
Hello everyone,

I would be surprised if the MMU does not support back-to-back load/stores (or instruction accesses).

On my souvenir, it does support back-to-back requests. This is under the assumption that requests hits on the TLBs (instruction or data). Otherwise, currently there is a single PTW, so a miss in the TLBs is blocking (a subsequent translation request needs to wait for the previous one to be served).

You probably have a configuration that prevents back-to-back requests. That is my guess.

Anyway back-to-back processing for both instructions and data are critical for performance. In the case of the CVA6, because of its in-order nature, being able to fetch an instruction every cycle is essential. And when you have a highly non-blocking data cache subsystem (as the HPDcache) is a waste of performance to not support back-to-back on the data interface (load/stores).

khandelwaltanuj, 2025-04-08:
Hi @cfuguet 

When I say back to back, it means that it finishes to serve 1st transaction and next cycle a new request is presented. In its current form it doesnt seem to be possible for LSU. It works for instruction fetch. 

In the case of LSU lsu_valid_o (response valid) signal remains high for 2 clock cycles as you may see in the screenshots that I paste in my previous messages, making it impossible to do back to back.

Here I have pasted the screen shot again for you, with the corresponding message: 

there is lsu_vaddr_i: 29D5713FCD followed by 15849e011F followed by 3D63FBE118.
As per my understanding of the waves, either 15849e011F does not have any response or has the same response as 29D5713FCD. It looks very suspicious to me. Except if you tell me that back to back requests are not allowed.

[image]

Let me know what do you think about it ? 

Regards
Tanuj

cfuguet, 2025-04-08:
They are allowed.

We need more signals to know why the load_store_unit (lsu) is asking for the same information twice. In your waveforms, the two cycles lsu_valid_o corresponds to a response to lsu_req vaddr=29D5713FCD. That is why, you have the same response twice.

For example it may be that the cache was not ready do accept the load/store request the first cycle. That is why the LSU kept the signal two cycles. The second cycle the cache accepts the request, and as the translation is valid, the lsu can continue with another request.

khandelwaltanuj, 2025-04-08:
Hi César,

I am sorry, I forgot to mention, in the MMU there is no ready signal. Neither for of req nor for response.  

Let me explain you how the protocol works. 

1. You insert lsu_req_i, you have to keep it high until the response arrives. 
2. As soon as the lsu_valid_o goes high, the req needs to be de-asserted. 
3. lsu_valid_o remains high for another clock cycle after lus_req_i is de asserted. 

For me every lsu_valid_o 1 is a valid response but that is not the case. It remains high for 2 clock cycles. During the second clock, you cannot send another request. 

I hope it is clear to you now? 

@AngelaGonzalezMarino please do not hesitate to correct me if I am wrong.

Regards
Tanuj

cfuguet, 2025-04-08:
Hi Tanuj,

Thank you for the explanation but I respectfully disagree :)

It is a pipelined protocol. The LSU valid (lsu_valid_o) is the response/acknowledgement for the request taken in the previous cycle (lsu_req_i). There is no need for lsu_req_i to go down between requests.

In your simulation, if lsu_valid_o remains set to 1 for 2 cycles is because it is responding to two different requests. It is just that both requests are on the same address, then the responses are the same.

khandelwaltanuj, 2025-04-09:
Hi César,

Non problem. Let me give it another try :) 
Before showing some more waves I would request you to see these line of codes, from 507 to 515. 
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/cva6_mmu/cva6_mmu.sv#L507

lsu_valid_o is driven by lsu_req_q which is nothing but lsu_req_i shifted by a cycle(output of a register). So lsu_req_i has to remain high  for the same request until a respons is sent. 

Here are couple of example with waves 

In the first one, I just apply lsu_req_i for a clock cycle. lsu_valid_o is never asserted except for the requests who have ptw_error signal high.  You can see req_port_o are memory accesses by PTW and req_port_i are responses. 

[image]

In the second example here for the same test, I keep lsu_req_i high until the response arrives. I get the responses with lsu_valid_o with response lasting 2 clock cycles. And I cannot send any request until the second clock of response, starting from lsu_req_i posedge. 
[image]

I hope, its clear for you. As per my understanding MMU is pipelined. It is just my understanding and I may be totally wrong. I would like to hear from @AngelaGonzalezMarino what does she thinks about it ? 

Thanks for your help and patience César. 

Regards
Tanuj Khandelwal

cfuguet, 2025-04-09:
As I was saying in the previous comment, it is pipelined in the case of TLB hit. It is not in case of miss.

khandelwaltanuj, 2025-04-09:
I have to verify how does it work when there are many hits that follows. In the case of HIT it usually replies in the same cycle. 

I will just come back to my initial question, if there is a TLB miss, in the second case in my comment above. I cannot send another lsu_req_i as soon as lsu_valid_o is recieved, as lsu_valid_o is inserted for 2 clock cycles.  And for me this is an issue except if protocol clearly states that it is not possible to send another request just after a TLB miss, we need to wait another for atleast one clock cycle.

khandelwaltanuj, 2025-04-09:
And my first example simply shows that it is not possible to have lsu_req_i just for 1 clock cycle, it doesnt work. It has to  be kept to 1 until response arrives

cfuguet, 2025-04-09:
Yes of course.

The MMU can be see itself as a pipeline with in order issuing. If one request is stalled (the response is not ready the next cycle because e.g. tlb miss), then that request needs to wait and it is not possible to send another request.

khandelwaltanuj, 2025-04-09:
So we agree that following scenerio that I discussed before, is not allowed as this the case of TLB miss and me need to wait a clock cycle. Making 15849e011F an illigal request.  If @AngelaGonzalezMarino agrees with this, I will close the issue. 

there is lsu_vaddr_i: 29D5713FCD followed by 15849e011F followed by 3D63FBE118.
As per my understanding of the waves, either 15849e011F does not have any response or has the same response as 29D5713FCD. It looks very suspicious to me. Except if you tell me that back to back requests are not allowed.

[image]

Thanks a lot César with your help.

Tanuj

AngelaGonzalezMarino, 2025-04-09:
Hi, yes I think this makes sense

You can check how the LSU is handling that, which confirms that this is the expected behaviour (sorry I had not looked at this for about a year and took a while to remember)

https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/load_unit.sv#L336C1-L338C45
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/load_unit.sv#L373C1-L376C46

Thanks for your support @cfuguet !

khandelwaltanuj, 2025-04-09:
ok
So i will do the summary : 
1. In the case of TLB miss, lsu_valid_o is asserted for 2 clock cycles in the case of successful translation.
2. In the case of TLB miss, it is now allowed to send another request (lsu_req_i =1) until the lsu_valid_o is high, making it impossible to do back to back request. It is required to wait atleast for one clock cycle after the response is recieved. 

I was wondering if it is possible to add these protocol requirements in the MMU document please ? 

Thanks and Regards
Tanuj Khandelwal

khandelwaltanuj, 2025-04-09:
I will be sharing these notes on mattermost so that everyone is aware of these requirements. 

Thanks again

cfuguet, 2025-04-09:
I do not agree with those requirements.... Maybe we can discuss about it in one of the CVA6 meetings.

The lsu_valid_o needs to be set for only 1 cycle in the case of completed translation. Not 2 cycles. Again, the MMU currently supports back-to-back requests for translation requests hitting in the TLB. I do not see why you want to change that.

If you ask for 2 cycles, you are increasing the latency of all memory requests of 1 cycle and dividing the throughput by 2.

cfuguet, 2025-04-09:
Sorry, I just saw that you mention in the case of TLB miss... but anyway even in the case of TLB miss, I do not see why you want to constraint the MMU to keep the response valid signal for 2 cycles.

cfuguet, 2025-04-09:
[image]

cfuguet, 2025-04-09:
This is my understanding of the waves.

1. the request 29d5... waits to be responded. There is a first response after some time.
2. there is a second request 29d5... at the same time of the response for the previous.
3. One cycle after there is the response for the second request.
4. During the second response, the requester does a new request on 1584..
5. One cycle after the requester abandon the request.

So, to me, the MMU is not keeping the valid signal 2 cycles because of a miss. It is keeping it 2 cycles to respond to two back-to-back  requests...

Putting a constraint on the MMU saying that it needs to keep the valid signal 2 cycles after a TLB miss is not correct in my opinion. The MMU is always behaving the same: it responds the cycle after the request (once it has the translation available).

I would say that the protocol is complex and very ad-hoc anyway:
- There is no ready signal, hence the requester needs to keep its signal until it gets a response. In the meantime, the MMU is actually working for that pending request, without actual informing the requester. When the MMU does not respond the cycle after is because of a TLB miss, and in the background it starts the PTW to fetch the translation (PTE).
- The requester can abandon a request.
- The protocol accepts back-to-back requests, because the requester can issue a new request the same cycle as it is receiving the response for the previous request.

AngelaGonzalezMarino, 2025-04-09:
Hi,

Thanks for the explanations, Cesar!

Tanuj, if you want to avoid the second cycle of valid (second response to second request which was not intended), I think you have to use the dtlb_hit_o. 

You can see it in the behaviour of load unit:

When the load unit gets the grant, it checks if there was a dtlb_hit or not. If there was, it continues to send tag.
If there wasn't (your case), it aborts the transaction, and then moves to wait translation

https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/load_unit.sv#L303C5-L304C41
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/load_unit.sv#L374C9-L381C38

During wait translation, when it sees the dtlb_hit, it goes again to wait_grant and, in this case, it will continue to send the tag.
https://github.com/openhwgroup/cva6/blob/b9da1d9e2d6d0a4952084d370d5be84ed4bcd921/core/load_unit.sv#L303C5-L304C41

So basically, you can know if a valid is coming or not with the dtlb_hit signal, and adapt your request accordingly. I think this should solve the 2 cycles issue you mentioned.

khandelwaltanuj, 2025-04-10:
Hi César, 

I understand your analysis. But before I am totally convinced I need to show a couple of more waves so that I am sure that it is working as intended. 

I am finally using lu_dtlb_hit_o as valid response but we can see in this wave that it remains high for very long, and lsu_valid_o is high only for 1 clock cycle. 
[image]

Here are waves from icache fetch. Which I assume uses the same tlbs. Here I do not have 2 responses(miss and hit) for a similar kind of scenerio. There is only one respone from miss tlb. I was expecting the same behaviour as in the case of LSU. 
[image]

@AngelaGonzalezMarino can you please confirm me if these 2 scenarios are correct, even if they behave a little differently. 
Thanks and Regards
Tanuj Khandelwal

khandelwaltanuj, 2025-04-10:
I will just add another comment, using lu_dtlb_hit_o doesnt work in the case there is a PTW error.

AngelaGonzalezMarino, 2025-04-10:
Hi,

lu_dtlb_hit_o will remain high while the hit exists, i.e. as long as your lsu_vaddr_i and asid_i match with an entry in the TLB, which is until you change the input. (We don't see asid in your waveform, but I assume you are changing it when you change lsu_vaddr_i). 
Now the valid_o is one cycle, which I think is what you were looking for.

The second scenario, I don't see any signal for miss, so I am not sure what you mean. 

Anyway, instruction and data interfaces are different, and I think this might be related to the previous caches used. @cfuguet, maybe you know more about the original reason for this difference?

Last, about ptw error, yes, the simplification of using dtlb_hit is not always valid, I guess that's why the load unit has a state machine to handle this interface.

khandelwaltanuj, 2025-04-11:
Hi Angela,

I am for the moment keeping ASID constant(I have open a new issue on that) and just changing the request addresses. I have added more itlb signals in the following image. Instruction interface is different because in LSU every signal coming out of DTLB is registered(a flop is added). And it is not the case of icache interface. 

[image]

Yeah, it not easy to make mmu working stand alone. Though with your and César's help I am able to make many necessary changes and run many seeds stand alone. And I really apréciate your help. 

Thanks and Regards
Tanuj Khandelwal

NicolasDerumigny, 2025-04-15:
> Anyway, instruction and data interfaces are different, and I think this might be related to the previous caches used. @cfuguet, maybe you know more about the original reason for this difference?

I looked a bit into it, and this is due to the fact that `mstatus.MPRV` allors instructions to be decoded using physical addresses while data (load / stores) uses virtualisation, hence a dual interface. In my opinion, this could be simplified with a single interface but not without refactoring.

khandelwaltanuj, 2025-04-16:
Hi 

I have created these 2 waveforms to show the difference between the 2 intefaces. The one from LSU is a little strange and causes a lot of issues if the response is an exception and there is not dtlb_hit. In the case of a valid response we have to wait to for dtlh_hit (asserted a cycle before the valid response) and in the case of some of the exception(not all) we have to wait to exception valid(when dtlh_bit is not asserted) that arrives at the same time as rsp_valid. And in some other cases of exceptions dtlb_hit as well as exception is asserted. So its a bit of messy to manage. Specially in the case of back to back transactions. 

[image]

[image]

khandelwaltanuj, 2025-06-10:
Hello, 

I am closing this issue. As FSM within the CVA6 manages this scenerion properly. 

Regards
Tanuj Khandelwal

closed 2025-06-10 by khandelwaltanuj