openhwfoundation/cva6 #2908: [TASK] CV32A60X functional coverage status

ASintzoff, 2025-04-09:
### Is there an existing CVA6 task for this?

- [x] I have searched the existing task issues

### Task Description

This issue describes not covered items in functional coverage.
The commit used for this description is [440062b](https://github.com/openhwgroup/cva6/commit/440062b9c0aa9b4dd67860c0fc817d5d1588e5fd).

- CV32A60X.traps.Machine Mode External Interrupt.002_instructions.000
Automatically Generated Cross Bins for cross_external_interrupt</span>
Uncovered bins

cp_group | cp_interrupt | COUNT | AT LEAST | NUMBER
-- | -- | -- | -- | --
[auto_ENV_GROUP] | [MACHINE_MODE_EXTERNAL_INTERRUPT] | 0 | 1 | 1

- CV32A60X.traps.Machine Mode Timer Interrupt.002_instructions.000
Automatically Generated Cross Bins for cross_timer_interrupt
Uncovered bins

cp_group | cp_interrupt | COUNT | AT LEAST | NUMBER
-- | -- | -- | -- | --
[auto_ENV_GROUP] | [MACHINE_MODE_TIMER_INTERRUPT] | 0 | 1 | 1

### Required Changes

- explain why these items are not covered

### Current Status

- 2 items not covered

### Description of Done

Remaining not covered items are justified

MikeOpenHWGroup, 2025-04-26:
Hi @AyoubJalali, what is the current status of this task?  I believe it will gate TRL-5 of the CV32A60X.

ASintzoff, 2025-04-28:
All not covered items are explained

closed 2025-04-28 by ASintzoff