openhwfoundation/cva6 #2919: [BUG] <title>type hpdcache_data_wort_t is used before its declaration.

RakeshpatilLB, 2025-04-12:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

[image]
got this error while loading the files in Jaspergold. Working on the latest cva6 version. Please help me in resolving this issue, also let me know if any additional details are required.
Thanks.

JeanRochCoulon, 2025-04-13:
JasperGold is not yet supported by CVA6, but fell free to contribute : if you know how to modify the scripts to fix this issue, you can create a PR.

RakeshpatilLB, 2025-04-15:
Thanks for your response. I am relatively new to this project , i will definitely try if i could explore this project well enough, but the error here seems to be related to the code right?

RakeshpatilLB, 2025-04-18:
An update from my side:
The error in this case seems to be that the parameters being passed to HPDCACHE_TYPEDEF_REQ_ATTR_T are not getting assigned. So I moved the content of HPDCACHE_TYPEDEF_REQ_ATTR_T directly to the file cva6_hpdcache_subsystem.sv. After doing this, the design loaded successfully. However, I encountered numerous warnings during the elaboration of the loaded design (I used cva6.sv as the top module for elaboration). 

I have attached the TCL scripts responsible for loading the design along with the warnings file. Please review these files and let me know if I have included all the required files.

Whoever is working on synthesis or formal verification might be able to address this issue. I am unsure whether my approach is correct, so I would appreciate your guidance.

[run_load_gate.txt](https://github.com/user-attachments/files/19813364/run_load_gate.txt)

[flistcva6.txt](https://github.com/user-attachments/files/19813367/flistcva6.txt)

[cv_hpdcache.txt](https://github.com/user-attachments/files/19813369/cv_hpdcache.txt)

warnings : [jasper.log](https://github.com/user-attachments/files/19813464/jasper.log)