openhwfoundation/cva6 #2947: PMP may allow all acces in Machine Mode even if configuration is locked

khandelwaltanuj, 2025-04-25:
Hi @ninolomata 

I am reading following line of code and I am wondering in the condition if we are missing a & |confi_i.locked == 0 (something like this)? 

https://github.com/openhwgroup/cva6/blob/c784fd9047354a19306053b2e58e7f01a9452808/core/pmp/src/pmp.sv#L67

I can imagine a scenerio where, we are in a machine mode, lock = 1 but match = 'h0. I am wondering in this case it will let everything pass ? 

Can you please take a look at it ?

Thanks and Regards
Tanuj Khandelwal

Gchauvon, 2025-04-28:
RISCV Privileged Spec states that : 

> if no PMP entry matches an M-mode access, the access succeeds. If no PMP entry matches an S-mode
> or U-mode access, but at least one PMP entry is implemented, the access fails.

In Machine Mode if the access is outside of defined PMP region, it will succeed. Locked parameter is not used in this case. 
I believe the line of code is correct here.

khandelwaltanuj, 2025-05-05:
Hi 

Thanks for the clarification. I will align my bench to this spec. 

Regards
Tanuj Khandelwal