openhwfoundation/cva6 #2950: README is too long

cathales, 2025-05-02:
I think the README is too long, newcomers get easily lost.
And new contents are still coming, such as #2949, so it gets longer and longer.

We should probably add a directory with one Markdown file per tutorial, and replace all the tutorials in the README with a short section with a list of hypertext links to the tutorials.

Proposition of new table of contents for README:

1. Presentation (merge Resources and Ecosystem section here)
2. Quick setup
3. Tutorials (links to other files)
   - Running simulations
   - ASIC Implementation
   - FPGA Implementation and running an OS
5. Directory Structure
6. Contributing
   - Contribute in the code => link to CONTRIBUTING + a link to the Kanban if it is still maintained for planned contributions
   - Contributing to the docs => Nothing?
   - Contributing with issues: merge Issues and Troubleshooting there
8. Fix the Publication part
9. Acknowledgements (should we inline this file instead of a link?)

Cc @jquevremont

MaxCThales, 2025-05-02:
I think this is a great idea. Splitting each tutorial into its own Markdown file will let us elaborate on individual topics in more detail, providing better support, without making the main README even longer.

jquevremont, 2025-05-07:
Excellent move! I'm very busy (with the Summit and upcoming vacation). Expect some delay to review this PR. Thanks

PR #2954 (2025-05-21): Readme refactoring
Fix #2950
commit c39333bd (2025-05-21): Readme refactoring (#2954)

Fix #2950

---------