openhwfoundation/cva6 #2955: [BUG] <title>PMP does not check misaligned accesses properly

davidharrishmc, 2025-05-05:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

I'm an outsider who has briefly looked at the pmp checking code and have a concern.

The PMP region-matching logic only relies on the physical address, not the length of the access.  This cannot be correct.  For example, consider fetching a 4-byte instruction aligned to an odd multiple of 2 bytes.  The address of the instruction could fall before the start of a region, but the upper two bytes could fall within the region, and the match logic would not detect this.

cvw has a similar bug that I am working to solve.

Moschn, 2025-05-05:
Hi,

I understand where you are coming from, and we hit exactly the same issues when integrating the PMP a few years back. Here is how we got around the issue: 

- We only support a minimum granularity of 16 or 8 bytes (not entirely sure which it is, depending on the 32-bit or 64-bit variant).
- The core does not support misaligned memory accesses. They must be aligned to multiples of 4 (or 8 in the 64-bit variant). Back in the day, we used m-mode software (such as bbl or openSBI) to translate any misaligned accesses into aligned accesses. Not sure how this is done in the current version.

These design decisions simplify the PMP region matching significantly, as we do not need to check access length, nor the other bytes that are accessed.

I hope this answers your question.

_Note:_ During the years, the design of CVA6 could have changed, though, so no guarantees that this is not an issue in the current version.

davidharrishmc, 2025-05-05:
> Moschn

I think you support compressed instructions.  How about a 4-byte instruction access to 0x8000_01FE, with a PMP region spanning 0x8000_0000 to 0x8000_01FF?

Worse yet, the instruction access should succeed if is a 2-byte access, and we don’t know its size until fetching at least the first two bytes.

Supporting only aligned load/store definitely reduces the number of corner cases.

davidharrishmc, 2025-05-05:
Maybe this problem could be avoided by setting a minimum granularity equal to the cache line size, such that it is necessarily split into multiple accesses?

Moschn, 2025-05-06:
Oh, I see the issue with instructions (especially compressed ones). Hmm, I do not remember if we thought of that at design time. 

To be honest, I have not looked into the RTL for a while, so I cannot tell you how (or if) the PMP checks are done for the frontend currently. I hope the core performs checks on entire cachelines, but not entirely sure. Maybe someone with more experience on the frontend can chime in?

OlivierBetschi, 2025-05-06:
Having a quick look at the RTL, on the data side the load_store_unit would detect a misaligned data and report it as a misaligned exception. For instructions, it seem to be that the instr_realign module would be able to cope with a full 32bits instruction on a 2 byte boundary but it do not see how an instruction (compressed or not) on an odd address would be taken care of.

davidharrishmc, 2025-05-06:
> OlivierBetschi

Looking at cva6/core/pmp/src/pmp_entry.sv:

        if (addr_i >= ({2'b0, conf_addr_prev_i} << 2) && addr_i < ({2'b0, conf_addr_i} << 2)) begin
          match_o = 1'b1;

Consider an instruction access from PC = 0x8000_0006, with a TOR PMP region spanning 0x8000_0000 to 0x8000_0007.

conf_addr_prev = 0x8000_0000
conf_adr_i = 0x8000_0004 (last two bits truncated)
addr_i = 0x8000_0006 (I believe, or maybe 0x8000_0004 if truncated)

This will set match_o = 1, no matter whether the instruction access is 2 or 4 bytes.

However, a 2-byte instruction should match and succeed, while a 4-byte instruction should match and fail.
But in pmp.sv, there is no check about the length of the instruction.  At line 60, allow = 1 so long as the access type is allowed.

Moschn, 2025-05-07:
> Looking at cva6/core/pmp/src/pmp_entry.sv:
>
>         if (addr_i >= ({2'b0, conf_addr_prev_i} << 2) && addr_i < ({2'b0, conf_addr_i} << 2)) begin
>           match_o = 1'b1;
>
> Consider an instruction [...]

Yes, the PMP check does not consider access length. However, as stated before, this does not matter in many cases:

- On a data access in the LSU (e.g., a load instruction): All accesses need to be aligned and the granularity is limited, therefore, the range check is not necessary in this case.
- On a PTW access: Pages are always aligned, therefore, the range check is not necessary.
- Instruction fetch: This is where I am not 100% sure how and where it is checked, and if the range check can be omitted.

davidharrishmc, 2025-05-09:
> fatimasaleem
>  left a comment
> (openhwgroup/cva6#2955)
> For instruction fetches, the CVA6 frontend always sends out 32-bit aligned addresses. When instruction data is returned, the frontend handles any necessary alignment adjustments internally if the original access was 16-bit aligned.
>
> From the PMP’s perspective, all fetch addresses appear as 32-bit aligned, and permission checks are performed based on 4-byte accesses.
>
> In cva6_icache.sv, the 32-bit aligned fetch_vaddr is sent to the MMU, where it undergoes address translation and PMP checks:
>
>   // make sure this is 32bit aligned
>   assign vaddr_d = (dreq_o.ready & dreq_i.req) ? dreq_i.vaddr : vaddr_q;
>   assign areq_o.fetch_vaddr = (vaddr_q >> CVA6Cfg.FETCH_ALIGN_BITS) << CVA6Cfg.FETCH_ALIGN_BITS;
> When the instruction data returns to the frontend, it is re-aligned based on the original (possibly 16-bit) request address:
>
>   // address will always be 16 bit aligned, make this explicit here
>   if (CVA6Cfg.RVC) begin : gen_shamt
>     assign shamt = icache_dreq_i.vaddr[$clog2(CVA6Cfg.INSTR_PER_FETCH):1];
>   end else begin
>     assign shamt = 1'b0;
>   end
>
>   // re-align the cache line
>   assign icache_data = icache_dreq_i.data >> {shamt, 4'b0};

This is interesting.  Does it fetch 32 or 64 bits from the 32-bit aligned address?

If it fetches 32 bits, does that mean that every 32-bit instruction on a halfword boundary requires two fetches?  That’s a common case and seems it would degrade performance.

If it fetches more than 32 bits, how does it check that the fetch doesn’t span PMP regions?

fatimasaleem, 2025-05-09:
> This is interesting.  Does it fetch 32 or 64 bits from the 32-bit aligned address?
> 

It fetches 32 bits.

But interestingly, CVA6 only supports a PMP granularity of 8 bytes, so, it wouldn’t be a problem even if it fetched 64 bits at a time.

in csr_regfile.sv
```
 // Important: we only support granularity 8 bytes (G=1)
          // -> last bit of pmpaddr must be set 0/1 based on the mode:
          // NA4, NAPOT: 1
          // TOR, OFF:   0
          if (pmpcfg_q[index].addr_mode[1] == 1'b1)
            csr_rdata = {pmpaddr_q[index][CVA6Cfg.PLEN-3:1], 1'b1};
          else csr_rdata = {pmpaddr_q[index][CVA6Cfg.PLEN-3:1], 1'b0};
```
> If it fetches 32 bits, does that mean that every 32-bit instruction on a halfword boundary requires two fetches?  That’s a common case and seems it would degrade performance.
> 
Yes, I agree that this case introduces performance degradation.

davidharrishmc, 2025-05-09:
Sounds like there is no PMP bug because every misaligned 32-bit instruction is decomposed into two aligned fetches.  

This seems very expensive because this 32-bit instructions make up at least 50% of a typical compressed instruction stream, and half will be misaligned, meaning a quarter of instructions require a spill that takes longer.  Wally solves this problem with a cache that supports misaligned accesses within the cache line.  It only needs to spill on instructions wrapping a 64-bit cache line.  This created a PMP bug for us, which was the original source of the bug report.  We’re working on solving the bug with a 64-B grain.

I’ll close this issue.

closed 2025-05-09 by davidharrishmc