openhwfoundation/cva6 #2963: Minor type decleration issue in ariane_xilinx.sv for the rvfi_probe_o

farnam-ponos, 2025-05-13:
https://github.com/openhwgroup/cva6/blob/806a12d16d196e3ddf8a46a34daeb0943c19207a/corev_apu/fpga/src/ariane_xilinx.sv#L205

Hi everyone, 

Probably, this is a minor issue, but working with `rvfi_probe_o` probe port at the `ariane_xilinx.sv` level, I noticed that the internal types of the struct type of `rvfi_probes_t`, i.e., csr and inst has not been properly set. It should be `rvfi_probes_csr_t` and `rvfi_probes_instr_t` for the `csr` and the `instr` respectively. Probably, the same issue in the parameter list of the ariane.sv itself.

Best 
Farnam

JeanRochCoulon, 2025-05-15:
@yanicasa do you confirm the issue ?
If yes, I invite you, @farnam-ponos, to submit a PR

yanicasa, 2025-05-17:
I understand it's confusing.
The RVFI port is not connected on the FPGA (line 770 of ariane_xilinx.sv).
In the CVA6 RVFI probes, declaring logic instead of the legitimate structure disables the probes to reduce the number of top-level ports. RVFI has many signals, and in a synthesized, non-instantiated CVA6, this creates output buffers that significantly increase the area.
When it's not connected, as in this case, either option can be used — it makes no difference.

farnam-ponos, 2025-05-19:
@JeanRochCoulon @yanicasa 
Thanks for clarifications, I guess I am fine, we can close this issue.

closed 2025-05-19 by farnam-ponos