openhwfoundation/cva6 #2974: [mmu question] Is it ok for the tlb_flush/flush to arrive during an ongoing transaction ?

khandelwaltanuj, 2025-05-20:
Hello

I have a question regarding the flush of MMU. In my UVM TB tlb_flush and flush may arrive anytime randomly. I get a feeling that when a tlb_flush arrives during an ongoing transaction, it may cause spurious memory access. Though I have been wondering if its really a valid use case ? 

Here is one scenerio in my waveform, where I am observing some spurious memory access after lsu_valid_o. 

[image]

Regards

cfuguet, 2025-05-20:
Hello @khandelwaltanuj,

After a quick read of the cva6/core/controller.sv file, it looks that this use case is valid. You can have both flush and flush_tlb valid at the same time.

Look into this [code snippet](https://github.com/openhwgroup/cva6/blob/2700d14471963909923bdcfa1b013b4d1b30e567/core/controller.sv#L162):

```verilog
    // ---------------------------------
    // SFENCE.VMA
    // ---------------------------------
    if (CVA6Cfg.RVS && sfence_vma_i) begin
      set_pc_commit_o        = 1'b1;
      flush_if_o             = 1'b1;
      flush_unissued_instr_o = 1'b1;
      flush_id_o             = 1'b1;
      flush_ex_o             = 1'b1;

      if (CVA6Cfg.RVH && v_i) flush_tlb_vvma_o = 1'b1;
      else flush_tlb_o = 1'b1;
    end
```

When committing a sfence.vma instructions, both signals are set to 1.

It should be easy to run a test executing the sfence.vma instruction to watch it in simulation.

khandelwaltanuj, 2025-05-21:
Thanks @cfuguet for your reply. 

My questions is more about, if flush can arrive during an ongoing translation ? When lsu_req_i is high and flush is inserted. 

Regards
Tanuj Khandelwal

cfuguet, 2025-05-21:
If you have the sfence.vma instruction immediately followed by a load instruction, I have the feeling that it is possible.

It would be interesting to run that unitary test.

khandelwaltanuj, 2025-05-21:
You can see the waves in my original message, its where tlb_flush inserted as lsu translation is ongoing. It seems to make some spurious memory accesses. 

My UVM bench is still not open sourced yet. Once it is done, I can provide you with the seed. 

Thanks

khandelwaltanuj, 2025-05-21:
If you think its not normal, I can mark it as a bug.

cfuguet, 2025-05-21:
As the target of this mmu is the CVA6, we should make sure that case can actually happen with the CVA6.

My feeling is that yes, with the example I gave before, but because of some buffering, may be it is not possible.

We should check first

khandelwaltanuj, 2025-05-21:
Ok,

In that case, we need to wait until I start the top level bench.

Thanks

closed 2025-06-26 by github-actions[bot]