openhwfoundation/cva6 #2988: [BUG] mmu seems to do spurious memory accesses in the case of misaligned exception

khandelwaltanuj, 2025-06-02:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hello, 

I am observing spurious memory accesses in the case of LSU access with misaligned excepiton valid. 

In the following figure, the yellow box presents the request and response from LSU agent and red box presents the memory access. 

[image]

It seems to come from this line of code of MMU. 
https://github.com/openhwgroup/cva6/blob/92b2d0b73b3ab825cbffcd4cc1430949b7d3f65d/core/cva6_mmu/cva6_mmu.sv#L173

Adding following condition seems to fix the issue. 
 assign dtlb_lu_access = lsu_req_i & !misaligned_ex_i.valid;

Following is the figure after the issue seems to be resolved. 
[image]

Thanks and Regards
Tanuj Khandelwal

AngelaGonzalezMarino, 2025-06-03:
Hi,

I am not sure if this is 100% correct. Apart from the differences you point out, in the first one there is a lsu_valid_o set to 1 after the memory request you want to avoid, (during lsu_paddr e13). Is this something you think is wrong too? 

In the second one this valid does not exist, because if you prevent dtlb_lu_access there is no hit, and so no valid, but I think you are also removing the exception information, which I think should be there. Can you show the exception output (lsu_exception_o) in both cases?

khandelwaltanuj, 2025-06-03:
Hi 

Oh I did not see that there is another valid_o with paddr e13. But that's strange. It means lsu_req_i (79298f4e13) has 2 responses ? 
One that arrives in the next cycle. And another with e13. ? 

I will see if I can reproduce the problem. And get back to you.

Thanks for looking into this.

khandelwaltanuj, 2025-06-03:
though my understanding of the design says: It should not performe translation when misaligned transaction is valid. By assigning dtlb_lu_access = lsu_req_i, it may do a PTW(thats how I understand this) if access are misaligned or not.

AngelaGonzalezMarino, 2025-06-03:
Thanks, how does the same look with your proposed change?

AngelaGonzalezMarino, 2025-06-03:
It seems the existing behaviour corresponds to what you wrote here

https://github.com/openhwgroup/cva6/pull/2989#issuecomment-2934640750

"Implementations that support misaligned accesses only to some physical addresses must translate and check
the address before determining whether the misaligned access may proceed, in which case
raising the page-fault exception or access is more appropriate"

The second exception valid in the existing implementation is giving a LD_ACCESS_FAULT

@fatimasaleem what do you think?

khandelwaltanuj, 2025-06-03:
I believe existing behaviour, have 2 responses. 1 after 1 clock cycle and 2nd after memory response is received. I will be very surprised if its normal to have 2 responses.

And LD_ACCESS_FAULT may comme from PMP checks or PTE checks.

khandelwaltanuj, 2025-06-03:
The code here seems to suggest that with misaligned accesses translation should not happen

https://github.com/openhwgroup/cva6/blob/92b2d0b73b3ab825cbffcd4cc1430949b7d3f65d/core/cva6_mmu/cva6_mmu.sv#L544

fatimasaleem, 2025-06-04:
Hi @khandelwaltanuj 

I agree with you — in the case of a misaligned exception, we should avoid sending a lookup request to the TLB.

Even though the current code ensures we never use the TLB output when a misaligned exception occurs,
```
if ((en_ld_st_translation_i || en_ld_st_g_translation_i) && !misaligned_ex_q.valid) begin 
```
but the TLB still performs the lookup in the background. If there’s a miss, it triggers a request to the PTW, which then sends a memory request — all of which is unnecessary, since we ultimately discard the result due to the exception. We should add a condition to skip the TLB lookup entirely in such cases to avoid this overhead.

On another note, this isn't a matter of exception priority — CVA6 already handles exception priorities correctly. My concern is about the second valid response from the MMU. The LSU performs multiple checks on the address, exceptions, and trans_id, so it's possible that the second response is always being dropped. This would mean we only act on the first response, and any following response (in the case of a misaligned exception, for example) is ignored, since the request is immediately dropped once the exception is flagged.

That said, we need to investigate further to understand why this hasn’t affected CVA6’s normal execution. But I strongly agree that we should add that condition on tlb lookup as you suggested.

khandelwaltanuj, 2025-06-05:
Thanks for looking in it @fatimasaleem. I believe we can agree that there is a bug here that need to be fixed as there are 2 responses. 

How do I generate GTKwaves ? I am using questasim.

fatimasaleem, 2025-06-10:
@khandelwaltanuj it will work if you can share the questasim wave file. 

I wouldn’t classify this as a bug, but it’s definitely a design optimization. To label it as a bug, we’d need to construct a test case that clearly fails due to this specific issue.

fatimasaleem, 2025-06-10:
Hi @khandelwaltanuj 

I was saying that we need to run a test at the CVA6 level and reproduce this bug. Currently you are running the test at the MMU level, but maybe we don't find the 2 responses issue at the core level, because we know that the scoreboard collects the responses based on the trans_id, and ofcouse we want to see the behavior of that load or store request that sends the MMU request. 

So if we somehow able to reproduce this issue at the core level, and see how the core reacts or will it be stuck in some FSM or not. That would be helpful for us to understand what is really going on.

fatimasaleem, 2025-06-11:
@khandelwaltanuj I understand your point, and I’m just curious how this issue hasn’t affected core functionality for so long. That’s why I wanted to review the waveforms for the specific test running at the core level.

Also, I never said we shouldn’t fix this—I fully agree that we should merge the fix you proposed. However, in parallel, I think it’s worth investigating why this went undetected for so long.

closed 2025-06-17 by khandelwaltanuj