openhwfoundation/cva6 #3007: [BUG] List index out of range error for performance model

ellial, 2025-06-12:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

When executing python3 perf-model/model.py verif/sim/<date>/spike_sim/hello_world.rv32imc.log (or any other core configuration), the following error comes up:

> line 635, in count_cycles
    start = min(e.cycle for e in retired[0].events)
IndexError: list index out of range

Commit: 2c881cb

The issue persists for different tests and configurations as well. I have checked that the selected log file is non-empty and generated correctly

JeanRochCoulon, 2025-06-12:
@cathales

cathales, 2025-06-13:
Hello!

Have you checked <https://github.com/openhwgroup/cva6/tree/master/perf-model#adapt-rvfi-trace-generation>?

cathales, 2025-06-13:
Hello again @ellial, can you try using #3008 and confirm that it works on your side?

Notice that the README has been modified as well, it should be easier now.

Here is the list of changes: https://github.com/openhwgroup/cva6/pull/3008/commits

Thanks for your feedback!