openhwfoundation/cva6 #3048: [BUG] <title>Error when compiling toolchain

LOCKEDGATE, 2025-07-09:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

I'm trying to compile the riscv toolchain, and have performed the following steps in sequence:

`sudo apt-get install autoconf automake autotools-dev curl git libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool bc zlib1g-dev`

`cd ./util/toolchain-builder`

`bash get-toolchain.sh`

`export INSTALL_DIR=/home/wjh/small_xiangshan/cva6/tool_chain_cva6`

`bash build-toolchain.sh $INSTALL_DIR`

When I was executing the third step, due to network problems, I failed to git the gcc repository. So I went to `https://github.com/gcc-mirror/gcc` to manually download and unzip it to the `src` directory

Then when I was executing the fourth step, an error occurred. 

[image]

I checked `./build/newlib-riscv-none-elf/riscv-none-elf/newlib/config.log`

[image]

I initially thought that the riscv-none-elf-cc in the Makefile was written incorrectly, so I changed it to riscv-none-elf-gcc
But then a new problem was reported

[image]

The gcc version of my virtual machine is 11.4.0. I don’t know what the reason for these problems is now, and I look forward to your reply.

LOCKEDGATE, 2025-07-09:
The error occurred in step 5, not step 4.

LOCKEDGATE, 2025-07-11:
Downloaded a new version of gcc, the problem is solved

closed 2025-07-11 by LOCKEDGATE

ericlin1231, 2026-03-17:
Hi, I encounter the same problem, although I install the new version gcc, there also has error that there is no riscv-none-elf-gcc, how did you fixed it ?