openhwfoundation/cva6 #3075: [hpdcache subsystem] Fetch issue when CVA6Cfg.AxiDataWidth < CVA6Cfg.ICACHE_LINE_WIDTH

khandelwaltanuj, 2025-08-14:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hello @cfuguet 

I am using following parameters: 

  localparam CVA6ConfigAxiDataWidth = 128;
  IcacheByteSize: unsigned'(32768),
  IcacheSetAssoc: unsigned'(8),
  IcacheLineWidth: unsigned'(512),

I have following scenario. 

I have a bootrom of word size 64 bits. It is connected to axi2rom which translate words from 64 bits -> 64 bits. 

After the data goes into an industrial NOC from 64 bits to 128 bits. Here NoC alignes the data according to the address. For ex in the case of addr 0x0, data is in 64 LSBits and in the case of addr 0x8 data is aligned to the MSB (127: 64 bits). 

When the data arrives in hpdcache subsystem. The entry data is 128 bits. Cache line is 512 bits.  It goes in the following code. 

https://github.com/openhwgroup/cva6/blob/bcb0f7de8654c8ad0b260d53a2ce070152d10c9f/core/cache_subsystem/cva6_hpdcache_subsystem_axi_arbiter.sv#L170C9-L170C57

At this line, I feel there is an issue, it doesnt chose anyword and I get instruction fault: 
https://github.com/openhwgroup/cva6/blob/bcb0f7de8654c8ad0b260d53a2ce070152d10c9f/core/cache_subsystem/cva6_hpdcache_subsystem_axi_arbiter.sv#L214

I was expecting a code something like this: 
https://github.com/openhwgroup/cva6/blob/bcb0f7de8654c8ad0b260d53a2ce070152d10c9f/core/cache_subsystem/cva6_hpdcache_subsystem_axi_arbiter.sv#L224

For the case of CVA6ConfigAxiDataWidth = 128, I have made following changes at line 214 (cva6_hpdcache_subsystem_axi_arbiter.sv) and it seems to work. 

      always_comb begin : icache_miss_resp_data_comb
        if (!icache_miss_req_rdata.mem_req_cacheable) begin
          automatic logic [ICACHE_CL_WORD_INDEX - 1:0] icache_miss_word_index;
          automatic logic [63:0] icache_miss_word;
          icache_miss_word_index = icache_miss_req_rdata.mem_req_addr[3];
          icache_miss_word = icache_miss_resp_data_rdata[icache_miss_word_index*64+:64];
          icache_miss_rdata = {{CVA6Cfg.ICACHE_LINE_WIDTH - 64{1'b0}}, icache_miss_word};
        end else begin
          icache_miss_rdata = icache_miss_resp_data_rdata;
        end
      end

Can you please comment on this ? 
Is it expecting data aligned to right in the case of CVA6Cfg.AxiDataWidth < CVA6Cfg.ICACHE_LINE_WIDTH 

Thanks and Regards
Tanuj Khandelwal

khandelwaltanuj, 2025-08-14:
And if I change parameters to following it works as AxiDataWidth = IcacheLineWidth 

localparam CVA6ConfigAxiDataWidth = 128;
IcacheByteSize: unsigned'(32768),
IcacheSetAssoc: unsigned'(32),
IcacheLineWidth: unsigned'(128),

cfuguet, 2025-08-21:
Hello @khandelwaltanuj,

I agree that in the case where AxiDataWidth < IcacheLineWidth uncacheable reads have an issue in some scenarios. I think we do not need to add additional code to fix it. We probably just need to relocate the code I've put in the case AxiDataWidth == IcacheLineWidth outside of the if-else. That way it will be used in both cases and uncacheable accesses shall be handled correctly.

I will test on my end and will propose a PR to fix it.

Thank you !