openhwfoundation/cva6 #3083: [BUG] Null instructions in dasm file

haipnh, 2025-08-22:
### Is there an existing CVA6 bug for this?

- [x] I have searched the existing bug issues

### Bug Description

Hi,
This issue exists on both master and v5.3.0 release when targeting `cv32a65x`:
```
Fri, 22 Aug 2025 12:34:56 INFO     [veri-testharness] Running ISS simulation: make veri-testharness target=cv32a65x variant=rv32imc_zba_zbb_zbs_zbc_zicsr_zifencei elf=/home/user/cva6/verif/sim/out_2025-08-22/directed_tests/cvxif_add_nop.o path_var=/home/user/cva6/ tool_path=/home/user/cva6/tools/spike/bin isscomp_opts="" issrun_opts="+enabled_cvxif +ntb_random_seed=101349881" isspostrun_opts="0x0000000080000000" log=/home/user/cva6/verif/sim/out_2025-08-22/veri-testharness_sim/cvxif_add_nop.cv32a65x.log &> /home/user/cva6/verif/sim/out_2025-08-22/veri-testharness_sim/cvxif_add_nop.cv32a65x.log.iss
```
Content of `trace_hart_0.dasm` looks like this:
```
                  73 0x10000 M (0x00000000) DASM(00000000)
                  73 0x10004 M (0x00000000) DASM(00000000)
                  78 0x10008 M (0x00000000) DASM(00000000)
                  79 0x1000c M (0x00000000) DASM(00000000)
                  83 0x10010 M (0x00000000) DASM(00000000)
                  83 0x10014 M (0x00000000) DASM(00000000)
                  94 0x80000000 M (0x00000000) DASM(00000000)
                  94 0x80000002 M (0x00000000) DASM(00000000)
                  95 0x80000004 M (0x00000000) DASM(00000000)
                  95 0x80000006 M (0x00000000) DASM(00000000)
...
```

Gchauvon, 2025-08-27:
It is the expected behaviour in cv32a65x to optimize area and power consumption.
The only supported tracer is the RVFI writing into `trace_rvfi_hart_00.dasm`. The other ones are legacy ones, not maintained for cv32a65x and cv32a60x.

haipnh, 2025-09-01:
Hi @Gchauvon , thank you for clarifying.
Our goal is to extract the cycle numbers from the .dasm log file. Is there a proper way to do this?

Gchauvon, 2025-09-08:
I guess, you will have to modify the tracer RTL.